raspberry-pi-camera/raspindi #3: Documentation

rudolfbyker, 2021-02-24:
Hi, would you add some documentation to help those who are new to NDI to get this up and running on their PI4?

rf152, 2021-02-24:
Hi Rudolf,

If you clone the repo and run build.sh, then it should build the binary under the build directory. You then need to copy that to somewhere in your path, and copy the libraries under Lib to `/usr/lib` or define LD_LIBRARY_PATH when running.

For example, if you copy the contents of the lib folders to `/opt/raspindi/lib` and the binary to `/opt/raspindi/bin`, then you would run the following command:

`LD_LIBRARY_PATH="/opt/raspindi/lib" /opt/raspindi/bin/raspindi`

rudolfbyker, 2021-03-09:
Besides `git`, what other dependencies do I need to install to get `build.sh` working?

```
$ bash build.sh 
src/main.cpp:17:10: fatal error: libconfig.h++: No such file or directory
 #include <libconfig.h++>
          ^~~~~~~~~~~~~~~
compilation terminated.
```

rudolfbyker, 2021-03-09:
OK, build works on vanilla Raspberry PI OS after `apt install libconfig++-dev`.

Next question: how should the folder structure of `/opt/raspindi/lib` look? No matter how I slice and dice it, I still get:

```
./raspindi: error while loading shared libraries: libndi.so.4: cannot open shared object file: No such file or directory
```

rudolfbyker, 2021-03-09:
OK, I'm starting to reverse-engineer this...

```
sudo apt install libjasper-runtime
```

Now it runs, but alas, it needs a config file, which is also not documented. Please assist :)

```
$ LD_LIBRARY_PATH="/opt/raspindi/lib" /opt/raspindi/bin/raspindi
Could not open config file /etc/raspindi.conf
```

After creating an empty file at `/etc/raspindi.conf`:

```
$ LD_LIBRARY_PATH="/opt/raspindi/lib" /opt/raspindi/bin/raspindi
terminate called after throwing an instance of 'libconfig::SettingNotFoundException'
  what():  SettingNotFoundException
Aborted
```

rf152, 2021-03-09:
Hi Rudolf,

If you look in the source tree under etc, there is a an example config file. (raspindi.conf.default)

If you are able to document the steps you took to compile and install, then I'll happily accept a pull request to improve the documentation.

rudolfbyker, 2021-03-09:
Thanks! I will write up everything I learn and submit a PR for the README.md file later. But only if I can get it working ;)

For now, more questions...

- How do I configure video settings like resolution, etc? The `raspivid` command line tool has a ton of configurable options: https://www.raspberrypi.org/documentation/usage/camera/raspicam/raspivid.md Run `raspivid` for a full list of options.
- Is `camera_number = "-1"` correct? What does `-1` mean here?

This is how far I get now:

```
$ LD_LIBRARY_PATH="/opt/raspindi/lib" /opt/raspindi/bin/raspindi
mmal: mmal_vc_port_enable: failed to enable port vc.ril.camera:out:1(BGR3): ENOSPC
mmal: mmal_port_enable: failed to enable port vc.ril.camera:out:1(BGR3)(0x1fc710) (ENOSPC)
camera video callback2 erroropen Failed to create camera component/home/pi/build/raspicam/src/private/private_impl.cpp 103
Error opening camera.
```

rf152, 2021-03-09:
To answer your questions:

* it doesn't have the ability to change settings on RaspiVid at the moment. It's locked to 720p (any higher and the frame rate drops).
* camera_number = -1 is fine. That is used for a separate part of software (https://github.com/raspberry-pi-camera/atem-tally)
* Shouldn't need running as root - it runs as the pi user on my cameras.

Have you got the camera module enabled (in raspiconfig)? And is it connected?

rudolfbyker, 2021-03-09:
Oops, I had another instance of raspivid running. After stopping it, `$ LD_LIBRARY_PATH="/opt/raspindi/lib" /opt/raspindi/bin/raspindi` runs with no output, so I assume it's working. Next step is to figure out the NDI client part. This is all new to me.

PR #5 (2021-03-20): Add beginner-friendly build and installation instructions
... and a few helper scripts :)

Closes #3
closed 2021-03-20 by rf152