raspberry-pi-camera/raspindi #8: Possibly broken by a recent Windows update?

rudolfbyker, 2021-05-12:
My RPI 4 running `raspindi` has been working well for a few months now, until about a week ago.

## Symptoms

`raspindi` source is detected, but video is just black. Other sources, like NDI's own screen capture, is still working.

## What changed?

The only thing that changed was a recent windows update, although I'm not sure which one. It stopped working on multiple Windows PCs, while continuing to work on other Windows PCs and all Linux PCs (using the exact same RPI4).

- The Windows PCs on which it does NOT work are all on update `10.0.19042`.
- The Windows PCs on which it DOES work are all on update `10.0.18363`.
- It's working on LinuxMint.

## Whose fault?

At first, I though this was a problem with the `obs-ndi` plugin that I'm using. See https://github.com/Palakis/obs-ndi/issues/651 . But NDI Studio Monitor is showing the exact same symptoms.

Since nothing changed in raspindi, it's probably not the `raspindi`'s fault.

However, since other NDI sources are still working, it might be possible to get this working again from `raspindi`'s side?

rudolfbyker, 2021-05-13:
Interestingly, there are a lot of recent reviews on the [NDI HX Camera](https://play.google.com/store/apps/details?id=com.newtek.ndi.hxcam) android app, reporting the same problem as above. This increases my belief that it might have to do with a Windows update. However, I can't find any official source from NDI or Windows talking about this problem.

closed 2021-05-17 by rf152

nanowireUK, 2021-10-29:
Pretty sure that whatever update affected Windows 10 is actually baked into Windows 11. I have exactly this problem as described on a clean Windows 11 install with updates.

rf152, 2021-11-01:
As I mentioned above, this issue seems to point to the Windows end.

NDI as a protocol is all handled by the (closed source) NDI library, so nothing has changed at that end.