raspberry-pi-camera/raspindi #9: GPU Support

cl0udb0y, 2021-05-17:
Hi - Really great implementation, works a treat and stable. I was wondering if you had looked at using the encoder h/w on the pi ? Taking a look at the stats it appears is all CPU and so not really possible to get a smooth stream  on a Model 3 above CIF (352 × 288) ?? The thought would be to use the mmal library and offload to the GPU/VPU?

rf152, 2021-05-17:
I haven't looked at this, although it looks interesting. I might have a look at the MMAL library and see if it can do any of the processing (which is pretty much just adding an alpha channel).

If you want to look yourself by all means fork and create a pull request, I'd be interested in seeing how well it performs if we can offload some of that processing.

rf152, 2021-05-17:
Having looked further into this, there is a newer version of the NDI library that appears to support Raspberry Pi specifically. 

I will have a look at this this week, and see if it works better.

rf152, 2021-05-18:
Ok, I've been looking at this, and got a little bit stuck down a rabbit hole. Keep your eyes open for v2.0.0, which completely rewrites how it retrieves the images from the camera, and from my limited testing so far works much, much better (smooth video even at 1920x1080).

closed 2021-05-18 by rf152

rf152, 2021-05-18:
Let me know how you get on with version 2.0.0; You should see an improvement on the framerate thing.

jgillamtmforum, 2021-05-18:
Great news thanks - I’ll test it out and report back.

jgillamtmforum, 2021-05-19:
@rf152 - works a treat at 1280x720 @ 30fps in a Model 3B+ - I have monitored the output at it holds up at 30fps, video is smooth. Searching around on monitoring GPU, seems impossible on the rPi. But a `pi@raspberrypi:~ $ sudo vcdbg reloc stats

'Relocatable heap version 4 found at 0x3c000000
total space allocated is 44M, with 44M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

small_allocs                  : 33
allocs                        : 173
alloc_fails                   : 0
legacy_block_fails            : 0
compactions                   : 0
discard_compactions           : 0
aggressive_compactions        : 0
aggressive_compaction_waits   : 0
aggressive_compaction_timeouts: 0
locks                         : 0
small_locks                   : 0'

Shows it's in use (before largest block is 235M bytes) which matched the rPi mem split

Slightly jerky at 1080p also noticed a warning `Aspect ratio forced to: 1920 x 1088` when the Conference file is set 
`neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="1080";
width="1920";
fps="30";`
free list at 0x3cfc1f00
15M free memory in 2 free block(s)
largest free block is 15M bytes`

I'll say again - awesome simple implementation of a raspberryPi studio camera. Thanks

rf152, 2021-05-19:
Yeah, the locked ratio is to do with the way that the camera captures; it has all sort of weird artifacts when you force it to 1920x1080. It's supposed to crop, but I couldn't get it to work properly. I might revisit it further down the line.