raspberry-pi-camera/raspindi #13: Support for PiZeroW - build opt. for ARM6

cl0udb0y, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
'illegal instruction' when running raspindi on piZeroW

**Describe the solution you'd like**
build opt. for PiZeroW to support addition Pi models

**Describe alternatives you've considered**
Not using PiZero : -)  - It's possible the single core won't be up to the task, but it stills has GPU and v2.0x is making full use to hopeful of 1280x720 @ 25fps is possible

**Additional context**

rf152, 2021-05-19:
This could be interesting. My initial thinking is I'm not sure how well the network would hold up, as it's wireless only, but it might work for a lower bandwidth stream.

My Pi Zero is in another project at the moment, but if/when I can free it up I'll have a look at this. I think the hardware is supported by NDI (which will be the limiting factor, as they're pre-packaged libraries). If you want to give it a go then you can download the embedded SDK and copy the correct library into the lib folder. (arm-rpi1-linux-gnueabihf).

jgillamtmforum, 2021-05-20:
Thanks Richard - I’ll try. On the network point I share the concern - for ref have m3b+ running with PoE (720p @ 30fps) and m4 over Wi-fi (1080p @ 30fps) both glitch free. Quietly confident the PiZero could handle the streaming. I’ll report back.

cl0udb0y, 2021-05-20:
Ok - replaced lib and rebuilt. Initial error gone, the app runs ( I see the NDI stream advertised) but on connection the Pi Zero borks with 'malloc(): corrupted top size' - did a little digging and this appears to be mmal not NDI. running 'sudo vcdbg dumpstate 2>/dev/null' results is this output
'2016638.086: Exception #1: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000000  r1:0000000a  r2:dff93e10  r3:76eace83  r4:7e0000c4  r5:00000001  r6:1f0d0744  r7:1f0d0744 
     r8:1ef92044  r9:1f0d0744 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef89cf0  lr:1ec638a0 usp:00000000 ssp:1ef89cf0  tp:1ef92044  sr:40000028  pc:1ec631b8 

2016638.126: Exception #2: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000006  r1:1ef24308  r2:00000000  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef89480  lr:1ec61914 usp:00000000 ssp:1ef89480  tp:1ef91ec4  sr:00000008  pc:1ec61906 

2016638.151: Exception #3: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000006  r1:1ef24308  r2:00000000  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef89490  lr:1ec61914 usp:00000000 ssp:1ef89490  tp:1ef91ec4  sr:00000008  pc:1ec61906 

2016638.172: Exception #4: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000006  r1:1ef24308  r2:00000000  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef894a0  lr:1ec61914 usp:00000000 ssp:1ef894a0  tp:1ef91ec4  sr:00000008  pc:1ec61906 

2016638.192: Exception #5: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000006  r1:1ef24308  r2:00000000  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef894b0  lr:1ec61914 usp:00000000 ssp:1ef894b0  tp:1ef91ec4  sr:00000008  pc:1ec61906 

2016638.215: Exception #6: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000006  r1:1ef24308  r2:00000000  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef894c0  lr:1ec61914 usp:00000000 ssp:1ef894c0  tp:1ef91ec4  sr:00000008  pc:1ec61c00 

2016638.236: Exception #7: illegal memory access - IPC, VC EMI, SYS EMI

  Registers:
     r0:00000000  r1:0000000a  r2:dff83928  r3:78337055  r4:7e0000c0  r5:00000001  r6:1f32b094  r7:1f32b094 
     r8:1ef91ec4  r9:1f32b094 r10:77777777 r11:77777777 r12:77777777 r13:77777777 r14:77777777 r15:77777777 
    r16:77777777 r17:77777777 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:1ef23320  sp:1ef894e4  lr:1ec638a0 usp:00000000 ssp:1ef894e4  tp:1ef91ec4  sr:40000008  pc:1ec631b8 

could not get scoreboard symbols
Pool 0x9f71f6b8 (1ef24560)
Malloc pool size=34K (pool=0x1f71f6b8-0x1f727f6c)
Bad segment links at 0x9f4a12b8
Relocatable heap version 4 found at 0x18000000
total space allocated is 108M, with 108M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

free list at 0x1eaef520
107M free memory in 1 free block(s)
largest free block is 107M bytes

0x18000000: free 98M
[... 1 lines cut ...]
0x1e2013e0: corrupt trailer (space 102781984 != 16449)
[  32] 0x1e205420: used 3.1K (refcount 0 lock count 0, size     3072, align   32, data 0x1e205440, d0rual) 'stabilise ref_ss'
0x1e206060: free 150K
[  21] 0x1e22b960: used  25K (refcount 0 lock count 0, size    25100, align    4, data 0x1e22b980, d0rual) 'ls table 4'
0x1e231bc0: free 171K
[  20] 0x1e25c6e0: used 4.3K (refcount 0 lock count 0, size     4378, align    4, data 0x1e25c700, d0rual) 'distortion table'
0x1e25c6e0: corrupt trailer (space 103143488 != 4449)
[  15] 0x1e25d840: used 135K (refcount 0 lock count 0, size   138240, align    4, data 0x1e25d860, d0rual) 'isp sw stage buffers'
0x1e25d840: corrupt trailer (space 103281792 != 138305)
[  14] 0x1e27f480: used 271K (refcount 0 lock count 0, size   277512, align    4, data 0x1e27f4a0, d0rual) 'concurrent mdata'
0x1e2c30e0: free 533K
[   6] 0x1e3486c0: used 6.1M (refcount 0 lock count 0, size  6422528, align 4096, data 0x1e349000, d1ruAl) 'cameraRIL:input pool'
0x1e9696e0: free 1.4M
[... cut ...]
[   2] 0x1eaef780: used 1.0M (refcount 1 lock count 8, size  1091584, align 4096, data 0x1eaf0000, d3rual) 'ARM FB'
[   1] 0x1ebfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x1ebfafc0, d0ruAl) 'audioplus_tmp_buf'
[   3] 0x1ebfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x1ebff000, d1rual) 'camera fast alloc arena'
heap corruption detected
small allocs not requested
could not get cp_isp_debug address' 

happy to keep poking, would need some guidance on where next .. thx
(btw Pi 4 at 2540 x 1440 @ 30fps works a treat over wifi)

rf152, 2021-05-22:
I would guess for that one you might be better going back to v1.1.1, which used the rapsicam library rather than mmal directly.

rf152, 2021-05-22:
I now have a Pi Zero W spare, so I will probably have a look in the week next week.

I'll let you know how I get on.

rf152, 2021-05-25:
I've had a look at this. It looks like the Pi Zero W just doesn't have enough memory to do this; the malloc error occurs in MMAL regardless. I don't have the programming knowledge to debug right into the MMAL code. 

For now, I think the Zero W is just unsupportable for this. If someone wants to dig further into it, feel free!

closed 2021-05-25 by rf152