raspberry-pi-camera/raspindi #14: cp: cannot stat 'lib/raspicam/*': No such file or directory Done

johannes1002003, 2021-05-21:
i get the message 
cp: cannot stat 'lib/raspicam/*': No such file or directory
Done

on a raspberry zero  with raspberry OS lite. 

after having executed  ./easy-setup.sh

what i could have done wrong ? 

Thanks a lot in advance 

Johannes

rf152, 2021-05-21:
That's a bug with the `install.sh` script, leftover from before the code used MMAL directly.

The Pi Zero is not technically supported, so you may run into library issues. #13 details cl0udb0y's attempts to get it working

johannes1002003, 2021-05-21:
Thanks a lot for your lightspeedish reply, Johannes

johannes1002003, 2021-05-21:
Hello Richard, maybe it would be helpful to mention the hardware expected and the Raspi OS Version (lite or complete) 

And if additional things should be installed like VLC or so. 

Thanks a lot , Johannes

johannes1002003, 2021-05-21:
Now i changed to a raspberry 3 B+ 
but i still get this error 

pi@raspberrypi:~/raspindi $ /opt/raspindi/raspindi.sh
Parse error at /etc/raspindi.conf:15 - syntax error

rf152, 2021-05-21:
Could you post the contents of your /etc/raspindi.conf file please?

johannes1002003, 2021-05-22:
neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="720";
width="1280";
fps="25";

# awb: "auto"; // Options: auto, sunlight, cloudy, shade, tungsten, fluorescent, incadescent, flash, horizon, max, off
# saturation: 0; // Value in range 0 - 100
# sharpness: 0; // Value in range 0 - 100
# contrast: 0; // Value in range 0 - 100
# brightness: 50; // Value in range 0 - 100
# exposuremode: "auto"; // Options: auto, night, nightpreview, backlight, spotlight, sports, snow, beach, verylong, fixedfps, antishake, fireworks, max, off
# meteringmode: "average"; // Options: average, spot, backlit, matrix, max
# rotation: 0; // Options: 0, 90, 180, 270
# mirror: "none"; // Options: none, horizontal, vertical, both

rf152, 2021-05-22:
Could you try just putting in:

```neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="720";
width="1280";
fps="25";```

And see if it runs with that?

johannes1002003, 2021-05-22:
Then this happens: 

***@***.***:~ $ /opt/raspindi/raspindi.sh

terminate called after throwing an instance of 'libconfig::SettingNotFoundException'

  what():  SettingNotFoundException

Aborted

 

As a shortcutting question: 

Which PI-OS version you use
Which Raspberry 

Something else installed like VLC ?

Thanks a lot for your help, I really want to get this working, 

 

Kind regards Johannes

rf152, 2021-05-22:
My apologies Johannes, I left one of the config options out. Try:

```
neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="720";
width="1280";
fps="25";
```

In terms of other software such as VLC, there are no requirements. If you've used the easy_setup script, then it will have installed all the required dependancies.

johannes1002003, 2021-05-23:
Hi bRichard, 

 

when deleting the „space“ and retyping space before and after the equal sign it works. Seems that is something else then ascii 32

 

it works now 

 

Thanks a lot for poking me in the right direction. 

Kind regards Johannes

closed 2021-05-23 by rf152