raspberry-pi-camera/raspindi #18: No video when viewing Raspberry Pi NDI source.

skarard, 2021-05-29:
**Describe the bug**
Connecting to the Raspberry Pi NDI source doesn't display an output. I don't receive an error.

**To Reproduce**
1. Run `/opt/raspindi/raspindi.sh`
2. Use Obs-NDI plugin (or NDI Tools Webcam Input)
3. View `RASPBERRY PI>Video Feed`

**Expected behavior**
Stream video coming from the Raspberry Pi

**Additional context**
There were no errors.
This is a fresh installation of the latest Raspberry Pi OS (kernel 5.10)
I installed Barrier (for remote mouse and keyboard)
Followed the tl;dr (`easy-setup.sh`) from the readme.

**Hardware**
Raspberry Pi 4
Raspberry Pi HQ Camera

skarard, 2021-05-31:
To confirm, I used the released binary from 2.0.1 and it worked without issue.

rf152, 2021-05-31:
How were you doing it beforehand? Had you compiled them manually?

The release binary is pre-compiled (I've compiled them on my own Pi). It may be that something is not compiling properly in your specific case.

Glad you've now got it working. I'll close this issue

closed 2021-05-31 by rf152