raspberry-pi-camera/raspindi #21: Discussion: Deterministic timing

rudolfbyker, 2021-06-15:
Is there a way to make the delay more deterministic? Maybe by timestamping the NDI stream somehow and then making sure all of my PCs are using the same NTP source?

I'm having the issue that I have to modify the delay on a weekly basis to get the RPI-NDI camera (on the same gigabit switch as the streaming PC) in sync with the audio, which is captured directly through an audio interface (not connected to the RPI). It is usually around 333ms, but this past Sunday it was more than 1s, while nothing changed in the hardware or software setup on either the RPI or the streaming PC.

rf152, 2021-06-17:
That's an interesting question. I'm not too sure how the NDI library handles timestamps, or whether there's a way of putting in a timecode or something. I'm not sure if that would help your issue, though, which seems to be related to latency on the network layer itself. Is there anything else on the gigabit switch other than the cameras and the PC that could be pushing traffic through?

Audio delay is one of the things that I struggle with too. Generally, I think mine stays pretty steady, but it can be a pain.

rudolfbyker, 2021-06-24:
I was hoping that, if the time on the RPI and the time on the PC are perfectly in sync (totally possible with NTP, or even with GPSD) then putting timestamps on the video frames which are offset sufficiently into the future to be much greater than the variance in the network latency (say, 2s) would allow the receiver to buffer it until such time when it's appropriate to use the appropriate frame exactly 2s after it was generated, regardless of network latency. Then I can set my audio delay to exactly 2s and call it a day.

But this is all based on theoretical assumptions of how NDI actually works, so I will have to do some more research.

rudolfbyker, 2021-06-26:
My suspicions are confirmed. This is both possible, and supported. We just need to figure out how to program it. This is from the "Introduction to NDI" PDF document that ships with the SDK:

> One huge advantage of NDI 4.5 capture is that all files recorded on a local system use its internal clock to provide frame sync. This means that you can capture multiple sources, even starting their recorders at different intervals, and easily sync them up for post-production using their embedded timecode.
>
> Better still, since the system clocks of multiple systems in different locations can easily be slaved to an online NTP (Network Time Protocol) server, numerous NDI recordings can match up with precision in non-linear editing systems for multi-cam workflows – even when these are captured at different venues!

[image]

rudolfbyker, 2021-06-26:
Note to self: Read chapter 14.2 of the SDK documentation on "TIMECODE SYNTHESIS".

APz, 2021-12-01:
Is there any progress on this? I'm looking into a streaming rig with multiple RPi high quality cameras and this would be the perfect software to run it.

rudolfbyker, 2021-12-01:
I found an interim "good-enough" solution, by setting "latency mode" to "low (experimental)" in the OBS-NDI plugin that I'm using. And then I forgot about this issue for a while. I have not made the time to do the necessary research yet. But it's technically possible according to what I have read thus far.

closed 2022-04-26 by github-actions[bot]