raspberry-pi-camera/raspindi #24: Lens Shading Calibration

IrPgFKS0, 2021-07-03:
I wanted to apply the changes described below for my arducam camera, but since raspindi doesn’t seem to use “raspstill”, the changes don’t take affect.  Is there a way to apply these changes via raspindi.conf file?

https://www.arducam.com/docs/cameras-for-raspberry-pi/native-raspberry-pi-cameras/lens-shading-calibration/

rf152, 2021-07-03:
From a quick look in the linked document, this would require a re-writing the code to apply the lens shading to the mmal capture object (see the section 2.5 Change the Raspistill.c file)

I don't have an arducam lens, and cannot test this. If you want to create a pull request to add this functionality then feel free.

IrPgFKS0, 2021-07-03:
I don’t know much about C, but I followed the linked instructions for my arducam camera and it worked like a charm, so maybe I will take a stab at it.

Do you know what “.c” object I would need to edit?

rf152, 2021-07-04:
The whole program is in `main.cpp`, so that's the file you'd need, yes.

The MMAL configuration is set between lines 440 and 485, so it would make most sense probably to put the configuration below that. If you want to have a config file option then you will need to add some logic to pull that out (see line 328 for an example of reading a config value).

yomebzh, 2021-08-04:
Hello, if you have found the solution, I am interested
  thank you

IrPgFKS0, 2024-03-20:
I just wanted to post here for other people that may have this question, that I did end up solving this issue.  Thanks to [Jörn Nettingsmeier](https://github.com/nettings) for the idea, I made another fork which allows you to simply apply the `--tuning-file` command directly to the `raspindi` binary and this for my Arducam worked perfectly... 

https://github.com/IrPgFKS0/raspindi

rf152, 2024-03-20:
There's some interesting work you've done. I'll try to have a proper look, and see if I can come up with a way of making it backward compatible with the config file format, and merge it into the main repo, if you're ok with that?

IrPgFKS0, 2024-03-21:
Yes, I am fine either way.  I mostly just wanted to post this solution so anyone could fix that shading issue (drove me nuts)!   Also the native libraries do support a config file flag so maybe that could be merged together with raspindi's config file.

**`rpicam-vid -h`**
`-c [ --config ] [=arg(=config.txt)]   Read the options from a file. If no filename is specified, default to 
                                        config.txt. In case of duplicate options, the ones provided on the command line
                                        will be used. Note that the config file must only contain the long form 
                                        options.`