raspberry-pi-camera/raspindi #25: How feasible is it to add Logitech (C920) Webcam support

jcoconnor, 2021-07-05:
Hi - newcomer to Raspi but long history in SW eng.

Seems this implementation uses mmal which is specific to the onboard camera.
Since I have a load of C920s spare I was wondering how feasible it would be for me to convert this over to use Webcam input.

Suspecting the answer is just go buy the onboard camera.

We are getting a pair of PTZ cameras for our livestreaming operation and was looking for a way to utilise the Webcams as local closeups for our worship band.

rf152, 2021-07-09:
Unfortunately with the way that the program works it pulls straight from the onboard camera. To add USB webcam support would require a complete rewrite of the application, adding significant complexity.

closed 2021-07-09 by rf152

jcoconnor, 2021-07-09:
@rf152 Thanks for the quick reply.
I had a look at the code and kind of guessed that might be the reply.
Suspect for the moment, purchasing an on-board camera is the way to go.