raspberry-pi-camera/raspindi #27: AWB Gain

heathbarpunch, 2021-07-05:
**Is your feature request related to a problem? Please describe.**
After successfully maintaining a stable stream on a pi4 w/HQ camera combo I started testing pi3B+ w/different camera modules I had on hand. Eventually I tested with a Lintest HD-1 hat and found that I could get a stable and performative stream but colors were unusable as in raspivid the HD-1 uses awb off, red Gain set to 1.0, and blue Gain set to 1.0, but gain control is not exposed in this project. 

**Describe the solution you'd like**
MMAL_PARAMETER_CUSTOM_AWB_GAINS available (preferably modifiable with the config file)

**Describe alternatives you've considered**
I'm still trying to figure this out myself but I'm not having much luck, so the only other alternative I have is that I am going to explore some other HDMI -> CS-2 modules as well but that's more avoid the problem then an alternative solution.

rf152, 2021-07-09:
Sounds like a good idea. I'll have a look at this hopefully this weekend.

rf152, 2022-03-01:
Ok, so it might be nearly a year on, but I have finally now implemented this in code! (sorry for the delay - life happened)

closed 2022-03-01 by rf152

heathbarpunch, 2022-03-04:
Thanks! Even a year out still beat my availability to work through it. I'm going to test the latest revision on my HQ sensor equipped Pi4 this weekend and I should be able to test the AWB settings on the hat I requested the change for in 2 weeks (I'm traveling and that particular unit doesn't travel well). If I see any notable oddities I'll open a new issue and let you know.