raspberry-pi-camera/raspindi #28: python command

yomebzh, 2021-07-27:
hello, is there the possibility to change the parameters of the camera, like the contrast or the color thanks to a python command ? before installing Raspindi I used the command
#### CONTRAST IMG -100 - 100 (step1)
    def imgSetContrast(self,m):
        if -100<=m<=100:
            subprocess.call(['v4l2-ctl', '--set-ctrl', f'contrast={m}'], shell=False)
            self.imgContrast=m
            print (f'Contrast={m}')
necessarily now it is no longer valid
thank you for your help and your work

rf152, 2021-07-27:
You can set these settings in the raspindi.conf file; you can see raspindi.conf.default for an example.

closed 2021-07-27 by rf152

yomebzh, 2021-07-27:
thank you for your answer, Yes of course, that's what I did,
 neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="1080";
width="1920";
fps="25";

awb: "auto"; // Options: auto, sunlight, cloudy, shade, tungsten, fluorescent, incadescent, flash, horizon, max, off
# saturation: 0; // Value in range 0 - 100
# sharpness: 0; // Value in range 0 - 100
contrast: 10; // Value in range 0 - 100
# brightness: 50; // Value in range 0 - 100
exposuremode: "auto"; // Options: auto, night, nightpreview, backlight, spotlight, sports, snow, beach, verylong, fixedfps, antishake, fireworks, max, off
# meteringmode: "average"; // Options: average, spot, backlit, matrix, max
# rotation: 0; // Options: 0, 90, 180, 270
# mirror: "none"; // Options: none, horizontal, vertical, both
# flickeravoid: "50hz"; // Options: auto, 50hz, 60hz, max, off

but once the file was compiled, no more direct modification possible, Or else I didn't understand

rf152, 2021-07-27:
The file is read on run; it’s read from /etc/raspindi.conf

yomebzh, 2021-07-27:
thanks a lot