raspberry-pi-camera/raspindi #30: `raspindi.sh` hangs on Raspberry Pi 3B

batchku, 2022-02-14:
**Describe the bug**
Running ` /opt/raspindi/raspindi.sh` hangs in the terminal; nothing happens.
This is on a Raspberry Pi 3B with a Pi Camera Module 1.2.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow installation instructinos based on Raspberry Pi OS; assure no errors
2. Run `/opt/raspindi/raspindi.sh`

**Expected behavior**
Something happens.

**Additional context**
Should this work with a pi camera module?

rf152, 2022-02-14:
When raspindi.sh is running do you see the NDI stream on the network? I wouldn't expect any output from the application itself.

batchku, 2022-02-14:
I do not see a stream.

Is it possible to use a Raspberry Pi Camera Module, as well as a simple USB camera?

batchku, 2022-02-14:
I discovered that the issue was with my RPi camera module; I tried a new one and the NDI stream is working well.

Two questions remain:
1. Can I use a USB webcam?
2. Where do I find the configuration parameters for this application (camera source, auto exposure, white balance, resolution, ...)

rf152, 2022-02-14:
For your two questions:

1. No, no plans to implement USB camera (see #20)
2. configuration parameters are in the raspindi.conf file, found at `/etc/raspindi.conf`

batchku, 2022-02-21:
Thank you; very helpful.

I

closed 2022-02-21 by batchku