raspberry-pi-camera/raspindi #31: Is it possible to run multiple pi's for multiple NDI streams?

batchku, 2022-02-21:
**Is your feature request related to a problem? Please describe.**
I'd like to run multiple pi's to have multiple video streams in the same network.

**Describe the solution you'd like**
Currently, when i try this, the first camera works but the second does not; the application runs but I don't see its NDI stream.  I suspect this is because the streams have the same name?

**Describe alternatives you've considered**
Add stream name to the configuration file?

**Additional context**
Add any other context or screenshots about the feature request here.

rf152, 2022-02-21:
The NDI stream takes the name of the host, so as long as your hosts are named different things (so not raspberrypi) then they should appear as separate streams.

batchku, 2022-02-21:
Aha, got it; the parenthetical in the NDI stream name is dynamically set; that does the trick, thank you.

closed 2022-02-21 by batchku