raspberry-pi-camera/raspindi #43: Is there a method to adjust camera settings while the NDI stream is running?

miniman13, 2022-04-08:
**Is your feature request related to a problem? Please describe.**
I would like to be able to adjust camera settings on the fly, is this possible? or do we need to change the config file and reload the service?

**Describe the solution you'd like**
Ability to adjust live, via control app, python interface, or some sort of web interface?

jaredwooden, 2022-04-09:
We built this super basic interface to make it as far as we could, but the only control exposed by _v4l2-ctl_ is focus_absolute. Control over the exposure, white balance, etc. would be priorities for us and happy to help if you can point us in the right direction. Thanks for all your work on this it works flawlessly

jaredwooden, 2022-04-09:
https://bitbucket.org/jaredwooden/rasppi-camera/src/master/

rf152, 2022-04-17:
This isn't something that I have a need for. If you'd like to code it and submit a pull request then feel free!

closed 2022-05-16 by github-actions[bot]