raspberry-pi-camera/raspindi #46: camera models applicable

wspeth, 2022-06-26:
could you just drop a word whether this solution is meant for raspberry camera modules or USB connected webcams also and how to configure?

heathbarpunch, 2022-07-08:
This is a library that either makes use of MMAL (if you are using legacy code) or libcamera if you are using the latest version. In both cases these libraries only work with the CSI port NOT with usb cameras.

 There is a solid half dozen issue pulls asking this question, so I strongly recommend reviewing closed issues before asking questions like this as you'll get answers faster.
 
For what you want to do: DiCaffeine will do that, or you can compile FFMPEG to do so for non-distributable purposes.

rudolfbyker, 2022-07-11:
> There is a solid half dozen issue pulls asking this question

True. This is a sign that it probably needs a very prominent place in the README.md for this repo. It's literally the most frequently asked question on here.

rudolfbyker, 2022-07-11:
See https://github.com/raspberry-pi-camera/raspindi/pull/47

rf152, 2022-07-11:
> Edit: Actually I just double checked that and libcamera does indeed see my C920, @rf152 does the switch to libcamera make usb integration more viable/possible?

To be honest I'm not sure. I have no plans to include USB webcam support, but if libcamera will make it work, then there's no reason it couldn't be included. If it's something that you wish to work on, then feel free!

closed 2022-07-11 by rf152