raspberry-pi-camera/raspindi #49: Segmentation Fault

stephen-hardy, 2022-09-28:
I have used the official imager to setup a new Pi 4 (4GB) running Buster lite three times: SD card 64bit, SSD 64bit, and SSD 32bit. After the machine comes up the first time, I apt update/upgrade, install git, and follow the install instructions given on this repo. For 64bit, I needed to do what was mentioned here:

https://github.com/raspberry-pi-camera/raspindi/issues/45

But, other than that, I think I'm working with clean installs that are free of any accidental misconfiguration.

The camera works. It has been working fine with a Pi 2, and I can use libcamera on the Pi 4 to pull images and video. Yet, I'm getting this every time I try to start raspindi:

`Options:
    verbose: 0
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 5000
    width: 1280
    height: 720
    output:
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: average
    exposure: auto
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 25
    denoise: off
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    bitrate: 0
    profile:
    level:
    intra: 0
    inline: 0
    save-pts:
    codec: YUV420
    quality (for MJPEG): 50
    keypress: 0
    signal: 0
    initial: record
    split: 0
    segment: 0
    circular: 0
Segmentation fault`

I've been around Linux and programming for many years, but I don't claim to be any great sage. To me, Segmentation Fault speaks of a memory management problem. So, I ran memory tests - which the device seemed to pass. I was initially trying to use a 64bit OS, and I know there are differences in how 32/64bit relate to memory, so I switched to 32bit. Didn't help, same "Segmentation fault". I wondered if maybe my somewhat cheap SSD was at fault, so I tried it on a standard SD card. Didn't help. 

I'm going to try a Pi 3, to hopefully prove whether my procedure can work. But, I still want to get a Pi 4 working. The readme references latency on a Pi 4, so someone has got it to work. Was that pre-Buster? What am I missing?

I hope I'm not missing something stupid. I've tried to do my homework to not waste anyone's time. But "Segmentation fault" isn't highly descriptive for a Google search. Can someone tell me how to get a more verbose output, at least?

Thank you!

stephen-hardy, 2022-10-02:
I just tried on a Raspberry Pi 3, and followed the instructions to the letter on a Buster lite 32bit OS. Freshly installed via the official Raspberry Pi installer. 

Got this:

`Parse error at /etc/raspindi.conf:17 - syntax error
Segmentation fault`

So... it is Lite specific? Or, it is Buster specific? It happens across architecture, storage medium type, Pi model.

I guess I'll try an older version, or full version, next.

monterolabs, 2022-10-17:
Hi,

I got same error, last version Raspbian on 3+ and 4

Thanks

stephen-hardy, 2022-10-18:
I was able to get it working, by using a binary from the releases. With all the people talking about their successful experiences using it in the Issues, I guess either everyone uses the releases, or I'm just especially incompetent. I could never make the Readme instructions work no matter what HW/OS combination or how carefully I followed them. I'm starting to wonder if those instructions represent someone's best recollection of installation, on an OS that has already had a bunch of other things installed. Or, perhaps it is a full desktop (never ended up getting to test that). I'll be shocked if anyone tells me those instructions work on a brand new headless image.

I have my notes on what worked on another machine. Tomorrow or the next day I'll get those posted here.

But, it is working working well on a Pi 4. About 215ms latency @ 1080p 30fps.

rf152, 2022-10-18:
Apologies for the delay on looking at this one. I've got no spare Pi boards at the moment (If only I could get my hands on one!), so can't test it. I'll see if I can free up one of my existing ones to try it out at some point.

Pleased you've managed with the binaries. My guess with a segfault is there's an issue with one of the libraries, but without a board I can't actually test it!

stephen-hardy, 2022-10-19:
OK, here is what I did. But, first, I'd like to call out that I made a mistake above. I referenced Buster when I should have said Bullseye. I thought Debian releases were unique to the letter. My mind held on to Buster for "B", and I thought I was good. Anyway...

- Raspberry Pi Official Imager
- Raspberry Pi OS Lite (32bit) - Debian Bullseye - Released: 2022-09-22
```
sudo apt update
sudo apt upgrade
sudo raspi-config
```
-     Performance Options -> GPU Memory -> 256
-     Interface options -> Legacy Camera -> Yes
```
wget https://github.com/raspberry-pi-camera/raspindi/archive/refs/tags/v2.0.2.zip
unzip v2.0.2.zip
cd raspindi-2.0.2
./easy-setup.sh
/opt/raspindi/raspindi.sh
```

And you get...
`Parse error at /etc/raspindi.conf:18 - syntax error`

`sudo nano /etc/raspindi.conf`
Go to the last line, delete carriage return and the close parenthesis, replace close parenthesis, CTRL-X, yes

WORKS! Now, here is what I did after it started working. THIS IS NOT ESSENTIAL, just an example of what also works.
### Configuration 1080p feed
`sudo nano /etc/raspindi.conf`
Change to the following:
```
neopixel_path = "/tmp/neopixel.state";
camera_number = "-1";
height="1088";
width="1920";
fps="30";
```

Using 1088 because 1080 results in "Aspect ratio forced to: 1920 x 1088"

### Setup as a service
`sudo nano /lib/systemd/system/raspindi.service`
Set content:
```
 [Unit]
 Description=RasPiNDI
 After=multi-user.target

 [Service]
 Type=idle
 ExecStart=/opt/raspindi/raspindi.sh

 [Install]
 WantedBy=multi-user.target
```

closed 2022-11-17 by github-actions[bot]

SBCV-apegram, 2022-12-07:
FWIW, the segfault happens on [line 162](https://github.com/raspberry-pi-camera/raspindi/blob/d0fc3721c34b4f7b8e058372819ef178cbce75a9/src/main.cpp#L162) of main.cpp. Commenting that out results in a bad_alloc on [line 134](https://github.com/raspberry-pi-camera/raspindi/blob/d0fc3721c34b4f7b8e058372819ef178cbce75a9/src/main.cpp#L134) of main.cpp. That's as far as I could get with my limited knowledge of C++.

Raspberry Pi OS Lite (64bit) - Debian Bullseye on a Raspberry Pi 4B (2GB) (I switched in the 64bit NDI modules as I mentioned in my other [issue](https://github.com/raspberry-pi-camera/raspindi/issues/45).

My original device (CM4 4GB/32GB) is still working flawlessly, I have not installed any updates or patches, and will not until this starts working...