raspberry-pi-camera/raspindi #60: Error while loading shared libraries: libcamera.so.0

daveyijzermans, 2023-02-02:
Hi! I wanted to try and get this working but I'm running into the following message when I try to run raspindi.sh:

`/opt/raspindi/bin/raspindi: error while loading shared libraries: libcamera.so.0: cannot open shared object file: No such file or directory`

I'm doing this on a currently fresh Raspbian Bullseye Lite install. I found [this](https://github.com/aler9/rtsp-simple-server/issues/1195) issue from another project that might suggest something has changed concerning libcamera in Raspbian. But I'm not skilled enough to look in your source and try to correct for it, if that is even the problem.

I've tried the precompiled files, easy-setup script, and step-by-step instructions each on clean installations but I continue to get to the same error message.

I might be overlooking something very basic here. Sorry if that's the case.

rf152, 2023-02-03:
Interesting, I'll have a look when I get a chance (might be tomorrow).

Interesting that it still has the issue when the software is built from scratch - judging by the issue you linked it is to do with libcamera re-naming their shared object library.

daveyijzermans, 2023-02-23:
> Interesting, I'll have a look when I get a chance (might be tomorrow).
> 
> Interesting that it still has the issue when the software is built from scratch - judging by the issue you linked it is to do with libcamera re-naming their shared object library.

Did you by any change manage to take a look? If not, can you tell be on which Raspbian version you can confirm this project working on? Then I could at least try it out for a bit. I'm looking to see if it will work with one of those HDMI to CSI adapters. Would make for a inexpensive NDI capture solution. Thanks!

rf152, 2023-02-24:
Sorry - work went crazy.

I've just done some more work on this one, so if you can pull the latest main commit, then you should be able to use the ./easy-install.sh script to build raspindi and it work (I've tested it on a Pi 4b running raspberry pi os lite 64-bit)