raspberry-pi-camera/raspindi #61: "Segmentation Fault" when trying to run

EFJackson, 2023-02-15:
Hi, really cool project and exactly what I need for something I'm working on. 

When trying to get this running on a fresh Bullseye install on a Pi 4 Model B, after following the step-by-step instructions, I get the following:

```
Parse error at /etc/raspindi.conf:17 - syntax error
Segmentation Error
```
The extent of my troubleshooting (hampered by a lack of skill) on this has been tracking down the reported line, which is:
```
# mirror: "none"; // Options: none, horizontal, vertical, both
```
I've tried the various listed options, as well as removing the line entirely and that hasn't done the trick.

I appreciate this is probably less a bug report and more a matter of my own inability, but any advice would be greatly appreciated nonetheless!

colinarelliott, 2023-02-16:
I am having the same issue as you with a Raspberry Pi 4B. I got rid of the config error on line 17 by removing the blank character on the last line. 

Now I'm just getting "Segmentation Fault", which I believe is some kind of memory error in compiling. Perhaps support for the model of Pi 4B we have is not yet complete?

```
root@cues-pi-cam:/opt/raspindi# . raspindi.sh 
Segmentation fault
```

EFJackson, 2023-02-17:
I know the project states it isn't guaranteed to support 4B, so I suppose I'm being a little cheeky from the get-go by asking anything. 

My experience was similar to yours, and that unspecific segmentation fault is where I gave up. I realise this is my own inability to patch past an unsupported Pi model however, I'm certain raspindi delivers admirably within its stated scope. Do let me know if you managed to find a workaround. 

I've resorted to using the much-less-flexible Dicaffeine in the meantime. It's pretty bulletproof for straightforward NDI streaming (though uses the legacy camera stack that raspindi does not), but judging from your profile we're in a similar line of work and you might find its reliance on a GUI (and therefore a mouse and screen) and 'program' behaviour as opposed to library less comforting for touring or set-and-forget integration into event/theatre tech. 

Cheers

rf152, 2023-02-17:
Thanks for reporting this.

Could you just confirm please:

- Which Raspberry Pi model are you using?
- Have you enabled the camera module on the device?
- What version of Raspbian are you using
- Have you tried the pre-compiled binaries?

The error with raspindi.conf is a known issue to do with the linebreak at the end of the file.

colinarelliott, 2023-02-17:
- Pi Model: 4B
- Enabled?: Yes, from the raspi-config cli command...
- Raspbian: 11

I have not tried the precompiled binaries. That will be one of my next steps

mayromlo, 2023-02-22:
Having Segfault issues as well. Hoping you can help. Fresh (today) install of Bullseye (32bit). All updates applied. Compiled from source.

-  Pi Model: 4B (4GB)
- Camera Disabled (default) : Segfault.
- Camera (Legacy) Enabled: "ERROR: the system appears to be configured for the legacy camera stack"
- Rasbian: Bullseye (11).....both 32bit (currently) and 64bit versions (yesterday)
- Pre-compiled (arch: arm7): "./raspindi-armv7: error while loading shared libraries: libcamera.so.0.0.2: cannot open shared object file: No such file or directory"

Confirmed Camera is up and running on the current 32 bit version of Bullseye using:
libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:8888
to stream to VLC over network.

rf152, 2023-02-22:
I have just managed to free up one of my Pi 4bs, so I'll have a look at this hopefully in the next few days.

rf152, 2023-02-22:
> Having Segfault issues as well. Hoping you can help. Fresh (today) install of Bullseye (32bit). All updates applied. Compiled from source.

Out of interest, why use 32-bit on a pi 4b?

mayromlo, 2023-02-22:
It's the default selected by raspberry pi imager. I normally change it to 64bit (and I did initially). Had no luck with it, so I thought it might be a 64vs32 thing (messing about with the precompiled libs). So I redid the 32bit image, but the same segfault there too. I don't think 32 or 64 matters much since I'm using a 4GB model (I think the distro is exactly the same). If I were to use a 8GB model then yeah I would definitely need 64bit to detect the extra RAM.

Thanks for looking into this.

mayromlo, 2023-02-23:
Not sure if this helps, but did manage to get a bit further.....still segfault at the end, but I get some output. I commented out "options->Parse(argc, argv);" in main.cpp (in src directory).....recompiled and now I get the options printed on the screen as follows (but it still segfaults):

pi@raspindi:~/raspindi $ /opt/raspindi/bin/raspindi
Options:
    verbose: 0
    info_text:
    timeout: 0
    width: 1280
    height: 720
    output:
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: average
    exposure: auto
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 25
    denoise: off
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file:
    lores-width: 0
    lores-height: 0
    autofocus-mode:
    autofocus-range:
    autofocus-speed:
    autofocus-window: all
    hdr: enabled232
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format:
    bitrate: 0
    profile:
    level:
    intra: 0
    inline: 0
    save-pts:
    codec: YUV420
    quality (for MJPEG): 0
    keypress: 0
    signal: 0
    initial:
    split: 0
    segment: 0
    circular: 0
Made it to after options Print
Running without preview window
Opening camera...
[31:16:18.940336210] [6589]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[31:16:18.984927833] [6597]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media3
Acquired camera /base/soc/i2c0mux/i2c@1/imx477@1a
Configuring video...
Mode selection:
    SRGGB10_CSI2P 1332x990 - Score: 4377.47
    SRGGB12_CSI2P 2028x1080 - Score: 2314.5
    SRGGB12_CSI2P 2028x1520 - Score: 3717.7
    SRGGB12_CSI2P 4056x3040 - Score: 34604.7
[31:16:18.994311726] [6589]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-YUV420 (1) 2028x1080-SBGGR12_CSI2P
[31:16:18.994585687] [6597]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [60..674181621]
    ScalerCrop : [(0, 0)/128x128..(0, 0)/4056x2160]
    AnalogueGain : [1.000000..22.260870]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [19989..674193371]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Video setup complete
Segmentation fault

Back to the drawing board :-)

rf152, 2023-02-24:
Ok, I've had a bit of time to look at this this morning, and it comes down to a breaking change in libcamera-apps, which has broken raspindi. I've now put in a fix (updated the header files that I include in this repo because libcamera-apps doesn't have a header package), which should resolve it for now.

I'll have to have a look at a more sustainable solution to this one, as I don't want to have to do this every time libcamera-apps changes something!

yomebzh, 2023-03-11:
hello, I have the same problem "default segmentation" on my new installation of cablecam

yomebzh, 2023-03-11:
pi@raspberrypi:~/raspindi $ /opt/raspindi/raspindi.sh
Options:
    verbose: 0
    info_text:
    timeout: 0
    width: 1280
    height: 720
    output: 
    post_process_file: 
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: average
    exposure: auto
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 25
    denoise: off
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: 
    lores-width: 0
    lores-height: 0
    autofocus-mode: 
    autofocus-range: 
    autofocus-speed: 
    autofocus-window: all
    mode: unspecified
    viewfinder-mode: unspecified
    metadata: 
    metadata-format: 
    bitrate: 0
    profile: 
    level:  
    intra: 0
    inline: 0
    save-pts: 
    codec: YUV420
    quality (for MJPEG): 0
    keypress: 0
    signal: 0
    initial: 
    split: 0
    segment: 0
    circular: 0
Running without preview window
Opening camera...
[0:20:53.385655425] [1873]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:20:53.461091230] [1881]  WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage!
[0:20:53.463060455] [1881]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
Acquired camera /base/soc/i2c0mux/i2c@1/imx219@10
Configuring video...
Mode selection:
    SRGGB10_CSI2P 640x480 - Score: 6093.33
    SRGGB10_CSI2P 1640x1232 - Score: 4557.83
    SRGGB10_CSI2P 1920x1080 - Score: 3250
    SRGGB10_CSI2P 3280x2464 - Score: 12887.7
    SRGGB8 640x480 - Score: 7093.33
    SRGGB8 1640x1232 - Score: 5557.83
    SRGGB8 1920x1080 - Score: 4250
    SRGGB8 3280x2464 - Score: 13887.7
[0:20:53.493007148] [1873]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-YUV420 (1) 1920x1080-SBGGR10_CSI2P
[0:20:53.493572935] [1881]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [75..11766829]
    ScalerCrop : [(680, 692)/64x64..(680, 692)/1920x1080]
    AnalogueGain : [1.000000..10.666667]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [21020..11767556]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Video setup complete
Opened NullEncoder
Requests created
Camera started!
Segmentation fault

marcelpoelstra, 2023-04-17:
Just to add some hopefully useful input

The last fixes from Feb24 indeed seem to fix the issues with Libcamera.
However, I tried installing a fresh RPi4 this weekend, using the RPi (Sony) HQ camera and it seems to output NDI, however, without the video.   Another breaking change in libcamera maybe?   
This setup turnsout  working fine using the old installation that has worked for over a year now (That SD card is gold now ;) )  
By the way, in the new setup, when using the libcamera-vid tool the camera itself works like expected.  Unfortunately It really seems something with  Raspindi again, just can't get my head around it myself...

rf152, 2023-04-17:
How are you installing raspindi? With the script, or with the pre-compiled binaries?

rf152, 2023-04-17:
I've just tested it with a fresh 64-bit lite image, installed using the easy-setup.sh script.

I get a working video stream with all the default settings.

marcelpoelstra, 2023-04-17:
The pre-compiled binaries do not seem to work in this situation  at all, they always come up with the segmentation fault like mentioned many times above.
So I tried thevfollowing use cases:
1)  Install a fresh (latest) Raspberry OS (Armv7) 64 bit.  Checkout git master, and run easyinstall.sh.
Outcome:  It's outputting NDI, but with empty video stream.  libcamera-vid sending UDP to my computer works like expected, so no issues with the camera or libcamera in that respect.
2) Install a fresh (latest) Raspberry OS armhf (32bit) 
Same steps, same outcome
3)Install the previous version of RaspberryOS (the one that was current when you made the fix)  
Same steps, same outcome.   Problem here is that libcamera gets updated as soon as you install libcamera-dev, so ends up being the same version as with the newest Raspberry OS distribution
4) Install the previous version of RaspberryOS (the one that was current when you made the fix)  
This time trying to prevent libcamera from updating by apt-mark hold libcamera ,  but this breaks things later in the process when libcamera-dev is installed, since that will be the newest version from the repo, and is not compatible with the older libcamera.

marcelpoelstra, 2023-04-17:
> I've just tested it with a fresh 64-bit lite image, installed using the easy-setup.sh script.
> 
> I get a working video stream with all the default settings.

That is strange, how come this doesn't work with me? 
Please note I use the HQ camera, but do I need to set something different in the .conf file maybe?
Never had to do that before to be honest

marcelpoelstra, 2023-04-17:
Bizarre.....   I will just start all over from scratch again,  but I really can't get my head around this.   And also, the old SD card with the image and this software from a year ago also still works flawlessly....

Sorry to bother you,  It must be some oversight here :S

rf152, 2023-04-17:
It's no problem - I'm keen to make sure it's not an issue with the software, as I want it to work for everyone as well as it seems to work for some!

It's an odd one, certainly. If you want I can image the sd card that I have and upload it so you can just image the SD card and know that it's a working raspindi, and you can then install any additional tools/software that you want to?

marcelpoelstra, 2023-04-17:
That would be great indeed :)   In the mean time I also will try again with a fresh install like you did

In any case I will post the result here.

marcelpoelstra, 2023-04-17:
OK to make sure, what I just did :

1) Download the 2023-02-21. from here : https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-02-22/

2) Use Raspberry Pi Imager 1.74 (Mac) to burn it on a microSD.

3) Boot the Pi with the fresh microSD

4) There is no git installed by default so I have to install that first :
apt install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  git-man liberror-perl
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man liberror-perl
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 7260 kB/7291 kB of archives.
After this operation, 37.7 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Ign:1 http://security.debian.org/debian-security bullseye-security/main arm64 git-man all 1:2.30.2-1+deb11u1
Err:2 http://security.debian.org/debian-security bullseye-security/main arm64 git arm64 1:2.30.2-1+deb11u1
  404  Not Found [IP: 199.232.150.132 80]
Err:1 http://security.debian.org/debian-security bullseye-security/main arm64 git-man all 1:2.30.2-1+deb11u1
  404  Not Found [IP: 199.232.150.132 80]
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/g/git/git-man_2.30.2-1%2bdeb11u1_all.deb  404  Not Found [IP: 199.232.150.132 80]
E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/g/git/git_2.30.2-1%2bdeb11u1_arm64.deb  404  Not Found [IP: 199.232.150.132 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Apt updat was needed before I could install git.

5.) clone the raspindi repo

6.) Run easy-setup.sh

marcelpoelstra, 2023-04-17:
Second attempt :

1.) Download the 2023-02-21. from here : https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-02-22/

2.) Use Raspberry Pi Imager 1.74 (Mac) to burn it on a microSD.

3.) Download source zip from github: 
wget https://github.com/raspberry-pi-camera/raspindi/archive/refs/heads/main.zip

4.) unzip 

5.) cd in to unzipped source dir :
cd raspindi-main 

6.) run ./easy-setup.sh

7.) 
........
.......
.......
Scanning dependencies of target ndioutput
[ 25%] Building CXX object src/CMakeFiles/ndioutput.dir/ndi_output.cpp.o
[ 50%] Linking CXX shared library libndioutput.so
[ 50%] Built target ndioutput
Scanning dependencies of target raspindi
[ 75%] Building CXX object src/CMakeFiles/raspindi.dir/main.cpp.o
[100%] Linking CXX executable raspindi
/usr/bin/ld: warning: libcamera.so.0.0.3, needed by /usr/lib/gcc/aarch64-linux-gnu/10/../../../aarch64-linux-gnu/libcamera_app.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libcamera-base.so.0.0.3, needed by /usr/lib/gcc/aarch64-linux-gnu/10/../../../aarch64-linux-gnu/libcamera_app.so, not found (try using -rpath or -rpath-link)
[100%] Built target raspindi
Done

So I'm afraid I'm really stuck here

rf152, 2023-04-17:
Interesting - thats an issue with finding the libcamera so, which is a new one on me. Can you try running `sudo apt install libcamera-apps`?

I'm just imaging the working SD at the moment, as soon as it's done, I'll upload it :)

marcelpoelstra, 2023-04-17:
Cool, I'm currently doing a new test, this time not using the official Raspberry Pi Imager app, but using BalenaEtcher on my Macbook, and doing the config edits on the boot partition of the microSD manually.
Somehow I want to make sure the official imager does not do any nasty tricks besides setting up Wifi and SSH.
Update on this coming up

marcelpoelstra, 2023-04-17:
OK, so I did the test again with the same image (23/02) and this time manually editing the ssh and wifi config on the bootfs partition of the microSD before putting it in the Rpi.
This of course went fine, however, the rest of the outcomes are the same.  
It ultimately failed on linking the executable again.

marcelpoelstra, 2023-04-17:
> Interesting - thats an issue with finding the libcamera so, which is a new one on me. Can you try running `sudo apt install libcamera-apps`?
> 
> I'm just imaging the working SD at the moment, as soon as it's done, I'll upload it :)

OK installing libcamera-apps did fix the linking issue.

However, I'm still getting  an empty NDI stream :

[image]
[image]

marcelpoelstra, 2023-04-17:
some more debugging :
When I tried to be smart and install libcamera-apps prior to running easy-install.sh, there is another linking issue popping up :

Scanning dependencies of target ndioutput
[ 25%] Building CXX object src/CMakeFiles/ndioutput.dir/ndi_output.cpp.o
[ 50%] Linking CXX shared library libndioutput.so
[ 50%] Built target ndioutput
Scanning dependencies of target raspindi
[ 75%] Building CXX object src/CMakeFiles/raspindi.dir/main.cpp.o
[100%] Linking CXX executable raspindi
/usr/bin/ld: warning: libcamera.so.0.0.3, needed by /usr/lib/gcc/aarch64-linux-gnu/10/../../../aarch64-linux-gnu/libcamera_app.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libcamera-base.so.0.0.3, needed by /usr/lib/gcc/aarch64-linux-gnu/10/../../../aarch64-linux-gnu/libcamera_app.so, not found (try using -rpath or -rpath-link)
[100%] Built target raspindi
Done

rf152, 2023-04-17:
I've now uploaded the raspberry pi image to https://github.com/raspberry-pi-camera/raspindi/releases/download/v3.0.3/raspindi-3.0.3.img.gz

Could you give it a go and see if it works as you expect it to.

Username is pi, and password is raspberry

marcelpoelstra, 2023-04-17:
On your first suggestion, after flashing a fresh microsd:   
sudo apt install libcamera0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libcamera0 is already the newest version (0~git20230124+9b860a66-1).
libcamera0 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

So I stopped here since that would not make any difference with my previous attempt where I first installed libcamera-apps before running easy-install.sh.

Now I'm up to the exiting part (hopefuly ;) )  and that is trying out your image.   Please hold on !

rf152, 2023-04-17:
Does it display if you use the NDI tools to monitor it (NDI Studio Monitor on Windows, NDI Video Monitor on Mac)?

closed 2023-04-17 by rf152