raspberry-pi-camera/raspindi #63: Build fails

CerberusVelvet, 2023-02-25:
I'm just trying this for the first time. I'm installing on a Pi4 Model B, running lightweight Pi OS (64 bit)

Following the 'compile your own' step by step instructions I get as far as "./build.sh" which generates the following output.

>

pi@ndipi1:~/raspindi $ ./build.sh
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/raspindi/build
[ 50%] Built target ndioutput
[ 75%] Linking CXX executable raspindi
/usr/bin/ld: warning: libavahi-client.so.3, needed by /home/pi/raspindi/src/../lib/ndi/libndi.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_entry_group_new'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_service_resolver_free'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_client_new'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_entry_group_commit'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_client_free'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_service_browser_new'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_entry_group_free'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_entry_group_add_service_strlst'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_service_resolver_new'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_client_errno'
/usr/bin/ld: /home/pi/raspindi/src/../lib/ndi/libndi.so: undefined reference to `avahi_service_browser_free'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/raspindi.dir/build.make:104: src/raspindi] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: src/CMakeFiles/raspindi.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

CerberusVelvet, 2023-02-25:
Resolved by:
sudo apt install libavahi-client-dev

rf152, 2023-03-21:
Thanks.

There are a few bits that are missing in the instructions, that's something I'm aware of. If/when I get time I'll write better instructions.

The easy_setup.sh script does do all the required setup for you.