raspberry-pi-camera/raspindi #64: Connected but no image

CerberusVelvet, 2023-02-25:
Clean install of Debian GNU/Linux 11 (bullseye) 64Bit on a Pi4B

Everything appears to be ok. I can grab a still image from the camera with `libcamera-still -o test.jpg`

raspindi appears to start cleanly. 
I can see the NDi source available in NDi Studio Monitor 
[image]

But I have no image at all, black screen.

What am I missing? 

```
pi@ndipi1:~ $ /opt/raspindi/raspindi.sh
Options:
    verbose: 0
    info_text:
    timeout: 0
    width: 1920
    height: 1080
    output:
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: average
    exposure: auto
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 25
    denoise: off
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file:
    lores-width: 0
    lores-height: 0
    autofocus-mode:
    autofocus-range:
    autofocus-speed:
    autofocus-window: all
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format:
    bitrate: 0
    profile:
    level:
    intra: 0
    inline: 0
    save-pts:
    codec: YUV420
    quality (for MJPEG): 0
    keypress: 0
    signal: 0
    initial:
    split: 0
    segment: 0
    circular: 0
Running without preview window
Opening camera...
[0:38:59.976511638] [2286]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:39:00.026632142] [2294]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring video...
Mode selection:
    SGBRG10_CSI2P 640x480 - Score: 8093.33
    SGBRG10_CSI2P 1296x972 - Score: 5797.33
    SGBRG10_CSI2P 1920x1080 - Score: 3000
    SGBRG10_CSI2P 2592x1944 - Score: 23450.2
Stream configuration adjusted
[0:39:00.037774787] [2286]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420 (1) 1920x1080-SGBRG10_CSI2P
[0:39:00.038101035] [2294]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [118..969249]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [32660..969367]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    Saturation : [0.000000..32.000000]
    AeMeteringMode : [0..3]
    Contrast : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    AwbMode : [0..7]
    AeConstraintMode : [0..3]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/1928x1080]
    AnalogueGain : [1.000000..63.937500]
    ColourGains : [0.000000..32.000000]
    Brightness : [-1.000000..1.000000]
    AeExposureMode : [0..3]
    AwbEnable : [false..true]
    Sharpness : [0.000000..16.000000]
Buffers allocated and mapped
Video setup complete
Opened NullEncoder
Requests created
Camera started!
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0

```

CerberusVelvet, 2023-02-25:
Sorry should have mentioned. Pi4B running latest raspberry pi os, lite, 64bit

Progress:
Still nothing from NDI Studio Monitor

Tried in OBS with the NDI plugin.

[image]

Seems to connect, send one frame, then stops. No further output in the PuTTY window and image remains still on that one frame in OBS.

rf152, 2023-03-21:
That's odd. I've just tested it on a blank Pi 3B+ and it works correctly.

I don't have a 4 to hand (oh to be able to get hold of a Pi), but I don't think it should have changed that much between the two. If I manage to free up one of my Pi4s then I'll give it a go on one of those.

rf152, 2023-03-21:
Ok, I've tested it with a Pi 4, and it's working as expected. I'm not sure what is wrong in your particular instance, but all I can recommend really is flashing a fresh copy of Raspberry Pi OS 64-bit lite to an SD card and trying again.

closed 2023-03-21 by rf152

chrisciborowski, 2023-05-27:
I know this issue is closed, but I have come across the same problem. Tried two installations of Bullseye Lite 64-bit on a Pi 4B. Raspindi sends a couple frames, then hangs. Any troubleshooting that you can recommend @rf152? Funny thing is the camera works fine with libcamera tools.

chrisciborowski, 2023-05-28:
Ok, back after testing. FWIW, 3B+ is also exhibiting the exact same behavior. Leading me to believe this is a camera issue. I have 2 cameras, both arducams based on the IMX708 camera module. Both function absolutely perfectly outside of using this tool (ie libcamera). Wondering if this is a camera module issue?

rf152, 2023-05-28:
I haven't tested it with other camera modules, as I don't have access to any of them. I don't think the code should care what the camera module is (it's just using standard calls). 

If you want to try playing around with the code to get it working and create a PR for it, you're welcome to.

chrisciborowski, 2023-05-28:
I picked up a HQ module with the IMX477 sensor, should be here tomorrow. Going to give that a shot with both setups and see if it works. If not, I'll dive deeper into this.

chrisciborowski, 2023-05-28:
Ran another test with the NDI Tools package and it works! Both on the Pi 4 and 3B+. That leaves me to believe there is an issue on the OBS side. Will research and update accordingly.

chrisciborowski, 2023-05-30:
Just for completeness sake, after upgrading OBS (29.1.2), obs-ndi (4.11.1) and the NDI runtime (5.5.4) on my windows streaming machine, removing and adding back the NDI source, and rebooting  both the Pi and the windows box, the Arducam IMX708 started woking in OBS. Super stoked!