raspberry-pi-camera/raspindi #65: Build errors

thekaibosh, 2023-08-24:
I used the latest raspbian bullseye 64bit image. I used the easy-setup.sh script, which installed several libraries (I've doublecheked that all of the compile and runtime libraries listed on the README are installed). when it executes the build.sh I get this compiler error:

-- Configuring done
-- Generating done
-- Build files have been written to: /home/camera/raspindi/build
Scanning dependencies of target ndioutput
[ 25%] Building CXX object src/CMakeFiles/ndioutput.dir/ndi_output.cpp.o
In file included from /home/camera/raspindi/src/../include/core/logging.hpp:1,
                 from /home/camera/raspindi/src/../include/core/options.hpp:22,
                 from /home/camera/raspindi/src/../include/core/video_options.hpp:14,
                 from /home/camera/raspindi/src/../include/output/output.hpp:14,
                 from /home/camera/raspindi/src/ndi_output.hpp:10,
                 from /home/camera/raspindi/src/ndi_output.cpp:8:
/home/camera/raspindi/src/../include/core/libcamera_app.hpp:55:33: error: ‘StreamRoles’ in namespace ‘libcamera’ does not name a type; did you mean ‘StreamRole’?
   55 |  using StreamRoles = libcamera::StreamRoles;
      |                                 ^~~~~~~~~~~
      |                                 StreamRole
make[2]: *** [src/CMakeFiles/ndioutput.dir/build.make:82: src/CMakeFiles/ndioutput.dir/ndi_output.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:115: src/CMakeFiles/ndioutput.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

Am I missing a library? Or do I not have a correct version?

jaybeeteee, 2023-08-29:
me too, exact same Error

JanLucaB, 2023-09-01:
I have exactly the same error.

I am using a Raspberry Pi 4 8GB.
I have already tested it with RaspberryOS Lite both 64- and 32-bit. I also tested it with the image from the release.

jaybeeteee, 2023-09-04:
hey, thanks for responding!
before you go (looking for and) setting up your pi station, would it make sense to install  aprevious version of libcamera?
how would I go about that?
apt-cache policy only lists the current version....

jaybeeteee, 2023-09-25:
any news on this?

rf152, 2023-10-04:
Please could you give this another go? I've checked it compiles, but I don't have a camera to hand, so I haven't been able to check that the app actually works!

jaybeeteee, 2023-10-04:
Hey Richard, no luck unfortunatley.

RaspberryPi 4, 8GB
RaspiOS 64bit Lite (latest, 2023-05-03)
clean install, apt install git, nothing else.

ndi@ndi-rpi1:~ $ sudo apt install git

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

The following additional packages will be installed:

  git-man liberror-perl

Suggested packages:

  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn

The following NEW packages will be installed:

  git git-man liberror-perl

0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.

Need to get 7282 kB of archives.

After this operation, 37.7 MB of additional disk space will be used.

Do you want to continue? [Y/n] y

Get:1 http://deb.debian.org/debian bullseye/main arm64 liberror-perl all 0.17029-1 [31.0 kB]

Get:2 http://deb.debian.org/debian bullseye/main arm64 git-man all 1:2.30.2-1+deb11u2 [1828 kB]

Get:3 http://deb.debian.org/debian bullseye/main arm64 git arm64 1:2.30.2-1+deb11u2 [5423 kB]

Fetched 7282 kB in 1s (8650 kB/s)

apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!

perl: warning: Setting locale failed.

perl: warning: Please check that your locale settings:

	LANGUAGE = (unset),

	LC_ALL = (unset),

	LC_TIME = "de_DE.UTF-8",

	LC_MONETARY = "de_DE.UTF-8",

	LC_ADDRESS = "de_DE.UTF-8",

	LC_TELEPHONE = "de_DE.UTF-8",

	LC_NAME = "de_DE.UTF-8",

	LC_MEASUREMENT = "de_DE.UTF-8",

	LC_IDENTIFICATION = "de_DE.UTF-8",

	LC_NUMERIC = "de_DE.UTF-8",

	LC_PAPER = "de_DE.UTF-8",

[... 450 lines cut ...]
/usr/bin/ld: main.cpp:(.text+0x840): undefined reference to `Output::OutputReady(void*, unsigned long, long, bool)'

/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraEncoder::createEncoder()':

main.cpp:(.text._ZN16LibcameraEncoder13createEncoderEv[_ZN16LibcameraEncoder13createEncoderEv]+0x94): undefined reference to `Encoder::Create(VideoOptions*, StreamInfo const&)'

/usr/bin/ld: libndioutput.so: undefined reference to `Output::timestampReady(long)'

/usr/bin/ld: libndioutput.so: undefined reference to `typeinfo for Output'

/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/10/../../../aarch64-linux-gnu/libcamera_app.so: undefined reference to `libcamera::Camera::generateConfiguration(std::vector<libcamera::StreamRole, std::allocator<libcamera::StreamRole> > const&)'

/usr/bin/ld: libndioutput.so: undefined reference to `Output::Output(VideoOptions const*)'

/usr/bin/ld: libndioutput.so: undefined reference to `Output::Signal()'

/usr/bin/ld: libndioutput.so: undefined reference to `Output::~Output()'

collect2: error: ld returned 1 exit status

make[2]: *** [src/CMakeFiles/raspindi.dir/build.make:104: src/raspindi] Error 1

make[1]: *** [CMakeFiles/Makefile2:142: src/CMakeFiles/raspindi.dir/all] Error 2

make: *** [Makefile:103: all] Error 2

jaybeeteee, 2023-10-04:
Richard, don't do anything,
I think I just got it, but I have to leave now, cannot test it.

I just ran "sudo apt upgrade", and then I was able to run the installer again, which completed successfully.
After that the /opt/raspindi/raspindi.sh script started without a bother.

I'll get the rest of my NDI setup here back online again tomorrow, and can report.

Thanks a million already,
ahoi, j

jaybeeteee, 2023-10-05:
OK Richard, this works!
I have no time to test properly, but the system is up and running, and it's fast!

I have immediately set ALL libcamera packages on the pi to "hold", so it should even survive future updates.....

I will come back here and report speeds/latencies once I've got the NDI environment set up again,
but for the time being this issue is solved, as far as I'm concerned.

Thanks again,
ahoi, j

rf152, 2023-10-07:
I'll close this issue now then. Fixed in #67

closed 2023-10-07 by rf152