raspberry-pi-camera/raspindi #66: Missing libencoders.so.1 library when running binary

thekaibosh, 2023-08-24:
I followed the steps on the README to run raspindi from the binary. I have a raspberry pi 4 with a fresh install of raspbian bullseye 64bit.

- ran `uname -m` and my arch is aarch64
- download libndioutput-aarch64.so into /usr/lib
- download libndi-aarch64.tar.gz and unzip into /usr/lib
- download raspindi.conf.default and mv to /etc/raspindi.conf
- download raspindi-aarch64 and mv to /usr/bin/raspindi. run chmod a+x
- run raspindi

result: 
root@camera2:/etc# raspindi
raspindi: error while loading shared libraries: libencoders.so.1: cannot open shared object file: No such file or directory

I've tried to search for a debian package that contains this file but I haven't had much luck

jaybeeteee, 2023-08-31:
Hey Ben,
did you get this working?

rf152, 2023-09-01:
From memory libencoders is part of ffmpeg. Have you got ffmpeg installed on the pi? (not sure what the name of the package would be, and I don't have a pi to hand at the moment)

JanLucaB, 2023-09-01:
Hey there,
I just tested it. 

I have installed ffmpeg (`apt install ffmpeg`). I still get the error:

`raspindi-aarch64: error loading shared libraries: libencoders.so.1: cannot open shared object file: No such file or directory
`

I also had to install "libconfig++9v5" (`apt install libconfig++9v5`) because I got this error before:

`raspindi-aarch64: error loading shared libraries: libconfig++.so.9: cannot open shared object file: No such file or directory`

XelaReko, 2023-09-06:
I am also getting the same error message and am unable to find the package that has this file. I built ffmpeg and libcamera just in case, but the file is still sadly missing.

XelaReko, 2023-10-05:
Last update fixed this.

rf152, 2023-10-07:
@ thekaibosh Are you able to test this after the latest update please?

closed 2023-11-05 by github-actions[bot]