raspberry-pi-camera/raspindi #69: raspindi stop a second after start with OBS plugin

monterolabs, 2023-10-08:
hi, I am trying to deploy two raspberry as cameras for a small church in the center of Madrid (spain), our idea is to broadcast the sunday by instagram, for this we have two raspberry with camera and a PC, in the pc we have obs last version with the plugin: https://github.com/obs-ndi/obs-ndi, before it worked fine but with the last versions we got errors compiling so now we move raspberrpi os to image (raspindi-3.0.3.img.gz) and raspberry works well but we start OBS...

./opt/raspindi/raspindi.sh
Buffers allocated and mapped
Video setup complete
Opened NullEncoder
Requests created
Camera started!
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 0 PVW: 0
PGM: 1 PVW: 1

thanks all

SBCV-apegram, 2023-10-10:
Can confirm, fresh install of RPi OS Buster (64-bit) and built from the master branch, the program freezes on a still as soon as the video source is marked as Preview or Program by tally.

SBCV-apegram, 2023-10-10:
I reverted to 25321ed Updated libcamera (#67) and recompiled, it works fine now. I guess it's still something with the tally/neopixel.

SBCV-apegram, 2023-10-11:
I submitted some comments on #68 that fixed the issue on my fork.

SBCV-apegram, 2023-10-11:
I submitted pull request #70

closed 2023-11-10 by github-actions[bot]