raspberry-pi-camera/raspindi #72: Build fails

sissoux, 2023-12-07:
I Tried to use the compiled version (on raspberry pi 4 running bookworm) but there is a lib compatibility error:

` ./raspindi-aarch64: error while loading shared libraries: libcamera_app.so.1.2.1: cannot open shared object file: No such file or directory`

Then I tried to build it as well and it seems there was again some breaking changes in libcamera. Or I'm doing something wrong ?

```
ln: failed to create symbolic link 'lib/lib-aarch64': File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/AlexPi/raspindi/raspindi/build
[ 50%] Built target ndioutput
[ 75%] Linking CXX executable raspindi
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `event_loop(LibcameraEncoder&)':
main.cpp:(.text+0x884): undefined reference to `LibcameraApp::OpenCamera()'
/usr/bin/ld: main.cpp:(.text+0x890): undefined reference to `LibcameraApp::ConfigureVideo(unsigned int)'
/usr/bin/ld: main.cpp:(.text+0x8a0): undefined reference to `LibcameraApp::StartCamera()'
/usr/bin/ld: main.cpp:(.text+0x8b4): undefined reference to `LibcameraApp::Wait()'
/usr/bin/ld: main.cpp:(.text+0x924): undefined reference to `LibcameraApp::VideoStream(StreamInfo*) const'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraApp::GetVerbosity()':
main.cpp:(.text._ZN12LibcameraApp12GetVerbosityEv[_ZN12LibcameraApp12GetVerbosityEv]+0x0): undefined reference to `LibcameraApp::verbosity'
/usr/bin/ld: main.cpp:(.text._ZN12LibcameraApp12GetVerbosityEv[_ZN12LibcameraApp12GetVerbosityEv]+0x4): undefined reference to `LibcameraApp::verbosity'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraEncoder::LibcameraEncoder()':
main.cpp:(.text._ZN16LibcameraEncoderC2Ev[_ZN16LibcameraEncoderC5Ev]+0x38): undefined reference to `LibcameraApp::LibcameraApp(std::unique_ptr<Options, std::default_delete<Options> >)'
/usr/bin/ld: main.cpp:(.text._ZN16LibcameraEncoderC2Ev[_ZN16LibcameraEncoderC5Ev]+0xcc): undefined reference to `LibcameraApp::~LibcameraApp()'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraEncoder::EncodeBuffer(std::shared_ptr<CompletedRequest>&, libcamera::Stream*)':
main.cpp:(.text._ZN16LibcameraEncoder12EncodeBufferERSt10shared_ptrI16CompletedRequestEPN9libcamera6StreamE[_ZN16LibcameraEncoder12EncodeBufferERSt10shared_ptrI16CompletedRequestEPN9libcamera6StreamE]+0x64): undefined reference to `LibcameraApp::GetStreamInfo(libcamera::Stream const*) const'
/usr/bin/ld: main.cpp:(.text._ZN16LibcameraEncoder12EncodeBufferERSt10shared_ptrI16CompletedRequestEPN9libcamera6StreamE[_ZN16LibcameraEncoder12EncodeBufferERSt10shared_ptrI16CompletedRequestEPN9libcamera6StreamE]+0x94): undefined reference to `LibcameraApp::Mmap(libcamera::FrameBuffer*) const'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraEncoder::createEncoder()':
main.cpp:(.text._ZN16LibcameraEncoder13createEncoderEv[_ZN16LibcameraEncoder13createEncoderEv]+0x20): undefined reference to `LibcameraApp::VideoStream(StreamInfo*) const'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o: in function `LibcameraEncoder::~LibcameraEncoder()':
main.cpp:(.text._ZN16LibcameraEncoderD2Ev[_ZN16LibcameraEncoderD5Ev]+0x50): undefined reference to `LibcameraApp::~LibcameraApp()'
/usr/bin/ld: CMakeFiles/raspindi.dir/main.cpp.o:(.data.rel.ro._ZTI16LibcameraEncoder[_ZTI16LibcameraEncoder]+0x10): undefined reference to `typeinfo for LibcameraApp'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/raspindi.dir/build.make:98: src/raspindi] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: src/CMakeFiles/raspindi.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

bgaster, 2023-12-15:
Is there an answer to this issue?

nettings, 2023-12-26:
Same here, latest bookworm 12.4. This is due to some renaming in libcamera - I guess they did not intend their libcamera-apps headers to be used by third parties. I'm not familiar with cmake, so I solved it with some horrible shotgun debugging.

1. Do not use the libcamera-apps headers shipped with raspindi, because they do not match the libraries that are getting linked later, causing the errors reported above. Instead, use the headers installed with the system:
```
cd include
rm -rf core/ encoder/ image/ output/ post_processing_stages/ preview/
ln -s /usr/include/libcamera-apps/* .
```
> This is not a proper solution, but I do not know how to sort out C++ headers with CMake.
> It should be possible to just include the system headers directly.

2. Edit src/main.cpp:
2.1 Replace `#include "core/libcamera_encoder.hpp"` with `#include "core/rpicam_encoder.hpp"`
2.1 Search and replace every occurrence of `LibcameraEncoder` with `RPiCamEncoder`.
3. Follow the normal build process.

nettings, 2023-12-26:
Runs for a while, but regularly segfaults every couple of minutes...
Looks like it happens here:
```
(gdb) bt
#0  0x0000007fb5ccaf64 in NDIlib_send_get_tally () from /opt/raspindi/lib/libndi.so.5
#1  0x0000007fb6211984 in NdiOutput::outputBuffer(void*, unsigned long, long, unsigned int) () from /opt/raspindi/lib/libndioutput.so
#2  0x0000007fb61c68b0 in Output::OutputReady(void*, unsigned long, long, bool) () from /lib/aarch64-linux-gnu/rpicam_app.so.1.4.1
#3  0x000000557156600c in void std::__invoke_impl<void, void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>(std::__invoke_memfun_deref, void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*&&, unsigned long&&, long&&, bool&&) ()
#4  0x0000005571565648 in std::__invoke_result<void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>::type std::__invoke<void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>(void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*&&, unsigned long&&, long&&, bool&&) ()
#5  0x00000055715646b8 in void std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>::__call<void, void*&&, unsigned long&&, long&&, bool&&, 0ul, 1ul, 2ul, 3ul, 4ul>(std::tuple<void*&&, unsigned long&&, long&&, bool&&>&&, std::_Index_tuple<0ul, 1ul, 2ul, 3ul, 4ul>) ()
#6  0x000000557156384c in void std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>::operator()<void*, unsigned long, long, bool, void>(void*&&, unsigned long&&, long&&, bool&&) ()
#7  0x0000005571561b60 in void std::__invoke_impl<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>(std::__invoke_other, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*&&, unsigned long&&, long&&, bool&&) ()
#8  0x000000557155f2d4 in std::enable_if<is_invocable_r_v<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>, void>::type std::__invoke_r<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>(std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*&&, unsigned long&&, long&&, bool&&) ()
#9  0x000000557155c7f0 in std::_Function_handler<void (void*, unsigned long, long, bool), std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)> >::_M_invoke(std::_Any_data const&, void*&&, unsigned long&&, long&&, bool&&) ()
#10 0x0000007fb61b7b24 in NullEncoder::outputThread() () from /lib/aarch64-linux-gnu/rpicam_app.so.1.4.1
#11 0x0000007fb592e9dc in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#12 0x0000007fb56eee18 in start_thread (arg=0x7fdfb44217) at ./nptl/pthread_create.c:442
#13 0x0000007fb5757e9c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
```

rf152, 2024-01-07:
This will be libcamera changing its api (again). Apologies for the delay in response, it was hectic up to Christmas, and then obviously there was Christmas and New Year!

The libcamera headers were not originally available as a package (hence including them in the source tree here). 

I'll try to review this properly this week!

closed 2024-02-05 by github-actions[bot]

bgaster, 2024-02-15:
any update on this?