raspberry-pi-camera/raspindi #74: Segfault after a few minutes of streaming

nettings, 2023-12-27:
**Describe the bug**
After a few minutes of streaming, the process terminates with a SIGSEGV. The build fixes from #72 were applied to the tree. This is on RaspiOS 12.4 Bookworm

**Additional context**
```
(gdb) bt
#0  0x0000007fb5ccaf64 in NDIlib_send_get_tally () from /opt/raspindi/lib/libndi.so.5
#1  0x0000007fb6211984 in NdiOutput::outputBuffer(void*, unsigned long, long, unsigned int) () from /opt/raspindi/lib/libndioutput.so
#2  0x0000007fb61c68b0 in Output::OutputReady(void*, unsigned long, long, bool) () from /lib/aarch64-linux-gnu/rpicam_app.so.1.4.1
#3  0x000000557156600c in void std::__invoke_impl<void, void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>(std::__invoke_memfun_deref, void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*&&, unsigned long&&, long&&, bool&&) ()
#4  0x0000005571565648 in std::__invoke_result<void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>::type std::__invoke<void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*, unsigned long, long, bool>(void (Output::*&)(void*, unsigned long, long, bool), Output*&, void*&&, unsigned long&&, long&&, bool&&) ()
#5  0x00000055715646b8 in void std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>::__call<void, void*&&, unsigned long&&, long&&, bool&&, 0ul, 1ul, 2ul, 3ul, 4ul>(std::tuple<void*&&, unsigned long&&, long&&, bool&&>&&, std::_Index_tuple<0ul, 1ul, 2ul, 3ul, 4ul>) ()
#6  0x000000557156384c in void std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>::operator()<void*, unsigned long, long, bool, void>(void*&&, unsigned long&&, long&&, bool&&) ()
#7  0x0000005571561b60 in void std::__invoke_impl<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>(std::__invoke_other, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*&&, unsigned long&&, long&&, bool&&) ()
#8  0x000000557155f2d4 in std::enable_if<is_invocable_r_v<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>, void>::type std::__invoke_r<void, std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*, unsigned long, long, bool>(std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)>&, void*&&, unsigned long&&, long&&, bool&&) ()
#9  0x000000557155c7f0 in std::_Function_handler<void (void*, unsigned long, long, bool), std::_Bind<void (Output::*(Output*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>))(void*, unsigned long, long, bool)> >::_M_invoke(std::_Any_data const&, void*&&, unsigned long&&, long&&, bool&&) ()
#10 0x0000007fb61b7b24 in NullEncoder::outputThread() () from /lib/aarch64-linux-gnu/rpicam_app.so.1.4.1
#11 0x0000007fb592e9dc in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#12 0x0000007fb56eee18 in start_thread (arg=0x7fdfb44217) at ./nptl/pthread_create.c:442
#13 0x0000007fb5757e9c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
```

nettings, 2023-12-27:
Applying the tally pointer fixes submitted by @SBCV-apegram (https://github.com/raspberry-pi-camera/raspindi/compare/main...SBCV-apegram:RasPi-NDI-HDMI:patch-1) fixed the issue for me.

rf152, 2024-01-15:
This merge request has now been approved, so this issue should be resolved.

closed 2024-01-15 by rf152